DeqingSun/ch55xduino #113: Smaller code size for core??

lpuser13, 2023-01-31:
Thanks for the great work on this repository... :)

Is it possible to make the base code size smaller.

For example I compiled a default/empty sketch for ch552 using CDC. The output shows:
`Sketch uses 4123 bytes (28%) of program storage space. Maximum is 14336 bytes.`

When I checked the generated .map file i saw the following:
[image]

Is there any way to change compiler optimizations to stop the highlighted or unused functions from being included?

DeqingSun, 2023-01-31:
You may have to edit the core to do the job. SDCC's linker is not very good at linking optimization

lpuser13, 2023-02-01:
ok, please close

serisman, 2023-02-12:
I noticed this a few months ago as well.  I have a project with a large buffer that wouldn't compile after I upgraded to the newest version.  It looks like the inclusion of UART1 took up enough extra RAM so my code wouldn't compile, even though I wasn't using either serial port.  (CH551 doesn't even have a UART1).

PR #115 is a potential solution to this that allows selection of which (if any) hardware serial ports to include in the build.