DeqingSun/ch55xduino #114: How to make macOs detect it when I use CH552 as a tiny keyboard?

nikejasonlyz, 2023-02-04:
Hi
Thanks for your maginificent work first.
I want to use CH552 to finish a 3 keys 'CV' keyboard for my girlfriend, writing the code reference to the example HidKeyboard on Windows/Linux, and it works.
When it connected to her macbook (m1), even the leds were anything but shiny. How should I make it work on macOS? Or is there anything wrong on PCB design?
Thanks a lot

DeqingSun, 2023-02-04:
please describe your problem and attach necessary code and circuit.

nikejasonlyz, 2023-02-05:
Problem: Her macbook (m1) doesn't seem to recognize the CH552. I don't know if it's because the circuit is underloaded or if the laptop lacks a certain driver. 😭The schematic is below.
[image]
Thanks for your help!

DeqingSun, 2023-02-05:
Use multimeter to test VCC voltage on the computer. It is very likely to be 0V. The problem is you left CC unconnected. You need 5.1K pull down for the host to detect it correctly. Refer to this design.

https://github.com/WeActStudio/WeActStudio.CH552CoreBoard/blob/master/HDK/WeAct-CH55xCoreBoard-V10%20SchDoc.pdf

nikejasonlyz, 2023-02-05:
(。・∀・)ノI Got it! Then it works well!

[image]

I‘ve never realized this problem because my computer only has the TypeA interfaces so I ignore the details. So in the previous learning design, the difference between 'TypeC' and 'Micro-B' in my mind was nothing but different about signal transmission rates, and directly replace the original MicroB circuit with TypeC. I was very grateful for your help! Have a great life!

closed 2023-02-05 by nikejasonlyz