DeqingSun/ch55xduino #117: Compiler breaks when creating and calling a function

XanCraft21, 2023-02-16:
Hello, i found another issue with the repo. When i create a function and use it, i always get problems. This does not seems to affect setup and begin, just any other function i make.

The errors are “implicit declaration” and “too many parameters”, i am declaring the function correctly with 2 parameters and calling the function correctly with 2 variables corresponding to each parameter.

I know this code only works in pure C so i looked up how to make the function, and it’s the same as c++. I tried moving the functions around and it makes no difference.

I am having lots of problems with computer hackers so it may be something to do with that, but if it isn’t can you see if you can fix it for me please? Thank you for any help you can give me.

DeqingSun, 2023-02-16:
please attach the code you got the bug.

XanCraft21, 2023-02-17:
I will send the code using gist of that's ok. Another error i forgot to mention is that it wants me to replace the return value on my begin function from "void" to "auto", but the function is not meant to return any values. I even tried moving the functions above my setup and loop functions but it made no difference.

The code i made is to drive a small display using SPI. The only library i used is the SPI library from the board package.

The code is here: https://gist.github.com/XanCraft21/ef3f6625192be81f2d3929bb9b25b795

DeqingSun, 2023-02-18:
Adding 

```
void begin();
void setWindow();
void sendData16(int DATA, bool DC);
void sendData(byte DATA, bool DC);
```

before setup will make your code work.

The problem is your code lacks function prototypes. Generally the .h files are used to specify the function prototypes. So the compiler know the function name, parameters, and return type when it is called in your code. Your code lacks those prototypes so the compiler has no idea how to call the function. 

It may be fine for GCC compiler to omit the function prototypes, and the similar code will compile for Arduino Uno. But SDCC is not that forgiving and you have to keep the prototypes, or your functions need to be declared in the right order. In your case, if you write "sendData" and "sendData16" first, then "begin" and "setWindow", and then "setup" and "loop". The compiler will know each function before it is called and the code can compile.

XanCraft21, 2023-02-19:
Ahh, i did not think to do that. I usually program in c++ and i have never tried pure c before. Thanks for the help, sorry if i wasted an issue space. I want to make sure i get it right before i close this.

XanCraft21, 2023-02-24:
That fixed the problem, thank you. But I discovered a new bug in the system, enabling an spi transaction does not work well. It sends bugged data to my display. The display i used changes offset when a bad command is send. Should i open a new issue for that?

DeqingSun, 2023-02-24:
Please add logic analyzer capture for reference. Or such bugs can not be verified

XanCraft21, 2023-02-28:
I do not have a logic analyzer, but during my tests i can tell the spi transactions are buggy by looking at how my display behaves with and without a transaction. I will do more testing and send a video if needed.

DeqingSun, 2023-02-28:
A video of buggy screen display contribute no value to the debugging process.

XanCraft21, 2023-03-01:
Ok. I will work on getting the stuff to test the bus output. I will close this issue since the original topic is solved. I will open a new issue once i get proof for a new bug. Thank you for your help.

closed 2023-03-01 by XanCraft21