DeqingSun/ch55xduino #121: Cannot send blovks of 64 bytes on USBSerial

buttim, 2023-02-28:
If I try to write a block of 64 bytes at once on ```USBSerial```, nothing is sent.
The same happens for blocks of 128 or 256 bytes, but not for 32 bytes.
The same thing happens if instead of using ```USBSerial_print_n``` I cycle for 64 times sending a single byte with ```USBSerial_write``` without flushing.

Sample code:
```
__xdata char s[256];

void setup() {
  memset(s,'x',256);
}

void loop() {
  delay(1000);
  USBSerial_print_n(s,64);
}
```

DeqingSun, 2023-02-28:
It works perfectly well on my computer.

[image]

Might be a compatibility issue? What OS, OS version, CH55xduino version do you use?

buttim, 2023-02-28:
Sorry for not specifying the details, I thought they were irrelevant in this case.
Windows 10
Arduino IDE 2.0
Ch55xDuino 0.014
I did not test with the serial monitor of the IDE but with Putty.
After your reply I tested more thoroughly with the serial monitor.
I discovered that the output appears but ~10 seconds later.

DeqingSun, 2023-02-28:
OK I reproduced the bug with my XP virtual machine. The code work for 63 and 65 but not 64. I'll test what was the cause.

buttim, 2023-02-28:
Thank you. I tried different variations, it does not happen if you flush outside of a 64 bytes boundary.
I gave a look at the source but could not spot any evident coding error

DeqingSun, 2023-02-28:
As far as I tested. In windows XP if I keep flushing data of 64 bytes, the data will not show up in the serial monitor, but if I flushing data of 64 then another length, they got printed. I'll investigate further later.

DeqingSun, 2023-03-01:
Some note during investigation:

The data did get sent to computer after each flush. 1st line is debug signal. 2nd and 3rd is USB.
[image]

And on my computer. For every 64 seconds, the 64*64 bytes got printed together. So it might be a descriptor issue? Need further testing.

buttim, 2023-03-01:
A really bizarre specification indeed. Let me know if I can be of any help, and hold on... I might have quite a lot of questions on the USB implementation coming... :)

DeqingSun, 2023-05-10:
https://github.com/DeqingSun/ch55xduino/commit/8f18a4f2f617334c3eff261343855f1390bc76f9

Should fix the issue.

closed 2023-05-12 by DeqingSun