DeqingSun/ch55xduino #122: Convert cryptic USB descriptors to more readable code

buttim, 2023-03-01:
In the examples the USB descriptors are mere arrays of bytes(```uint8t[]```).
Most of the bytes are just _magic numbers_ often with a comment explaining the meaning, others with no explanation at all.
Obviously such code is not easily understandable by a novice, let alone modify it.

I propose to adopt definitions from [the LUFA library](https://github.com/abcminiuser/lufa).

Converting all the examples would probably be overkill, but I think that starting with just one should be doable. I could collaborate on this. We might either convert one of the examples or add something new. I would be particularly interested in the possibility of supporting multiple CDCs, or mass storage + CDC

DeqingSun, 2023-05-26:
Lufa is copyright protected code and there need to be another source to copy the code.

prosper00, 2023-05-26:
from their website:
"LUFA is free to use - even for commerical purposes, subject to the MIT license restrictions (see project documentation)"

the license itself is here:
https://github.com/abcminiuser/lufa/blob/master/LUFA/License.txt
```
Permission to use, copy, modify, and distribute this software
and its documentation for any purpose is hereby granted without
fee, provided that the above copyright notice appear in all
copies and that both that the copyright notice and this
permission notice and warranty disclaimer appear in supporting
documentation, 
```

I'm no lawyer, but I don't think ch55xduino would be considered commercial. The commercial license adds the following features, none of which seem necessary for this project:
```
   1. The right to use LUFA without attribution
   2. Three free hours of consultation
   3. Priority for support requests
```

DeqingSun, 2023-05-26:
@prosper00 thanks for sharing. Seems reasonable. I'll fix the upload tool problem on my machine, before doing this job which take a lot of firmware write to test.

DeqingSun, 2023-06-02:
Most example converted.

closed 2023-06-02 by DeqingSun

buttim, 2023-06-02:
Great work indeed!