DeqingSun/ch55xduino #124: Problems with uploading via USB, no port on board. Bad bootloader?

XanCraft21, 2023-03-04:
Hello, this may not be a good issue but i need help on this. I tried using the USB port on the board to upload for the first time, and it did not work. I let windows install a driver, i jumped through loops and hoops to get it installed, it was installed fully and correctly, but the board never gave me an active port in Arduino.

It gives me a port for normal running mode without a driver, but nothing when i want to program it. Even though the library send the signal to activate program mode my board is basically junk without an external converter.

As far as i can see it is a problem with my board. In my past experience with USB based boards it doesn’t matter if i have a driver or not, if the board has no configured port due to bad bootloader design then it is not possible to program it without external hardware.

do you know how i can manually configure the board to have a port and device ID using the external hardware without needing to download any extra tools or software? If i need to reflash the bootloader i will, but there isn’t any other bootloaders online that might work, and so far nothing that shows anything about problems and how to fix them.

I am using Windows 10, the latest version of your library, and the latest version of legacy arduino 1.18.9.

Thank you for any help you can give me.

DeqingSun, 2023-03-04:
If your bootloader is corrupted, there is no way to fix it by yourself. The external programmer is not available to retail user including me.

Just change a chip and see how it works. It may also be a board design issue.

XanCraft21, 2023-03-04:
Ok, thank you. I will get a new chip when i can. Also what i meant by external programmer is just a serial TTL converter.

DeqingSun, 2023-03-04:
A serial adaptor will only work when the bootloader is functional.

XanCraft21, 2023-03-06:
Hmm, i see that. Also do you know how i can fix the upload port problem i am having? How do I manually configure a device port and signature on my bootloader without the need to download other tools?

DeqingSun, 2023-03-06:
I have no idea what you are talking about

XanCraft21, 2023-03-06:
I need to find an easy way to configure a usb port on my ch552 board within the bootloader. I’ve studied about this a little, but i don’t know how to modify the bootloader code.

DeqingSun, 2023-03-06:
I have no idea what is "configure a usb port" and I already told you that you can not "modify the bootloader code".

Pull D+ to vcc with a resistor while power up the chip will let the chip to enter the bootloader no matter what firmware it has. Note if you have a relative new chip it will only stay there for a few seconds.

closed 2023-03-07 by XanCraft21