DeqingSun/ch55xduino #125: [Feature request] Configurable bootloader access method

biemster, 2023-03-09:
My ch552g came in a macropad, with bootloader access method configured to pulling P1.5 to GND, instead of the standard pull P3.6 high. After using ch55xduino on it (which is awesome!), I tried the HID keyboard example, not realizing that the serial port would disappear and I would need to use the "first time programming" method again to reprogram it.
To my surprise the bootloader access method changed now to "Pull P3.6 high".

Can I configure this in ch55xduino? I prefer the "pull P1.5 to GND" method, but every time I program a sketch it seems to change to the default method.

DeqingSun, 2023-03-09:
It is hard coded in the upload software. For now, you may comment out the configuration part

https://github.com/DeqingSun/vnproch551/blob/cf9e2035a1dcc8145db77af51872ac1c767c8823/main.cpp#L507

Recompile and replace the tool in your Arduino environment.

biemster, 2023-03-10:
Thanks! That's very helpful. For my chip that is now already on the default `WriteOptionCmd`, do you maybe know which byte to change here:
https://github.com/DeqingSun/vnproch551/blob/cf9e2035a1dcc8145db77af51872ac1c767c8823/main.cpp#L487
to switch it to the P1.5 to GND method?

DeqingSun, 2023-03-10:
I would suggest just bypass WriteOptionCmd and use the official software to restore the configuration. So you avoid the risk of bricking your chip.

biemster, 2023-03-10:
Makes sense. Thanks!

biemster, 2023-03-12:
I'm reopening this, because I'm confused with the second `A8` command here:
https://github.com/DeqingSun/vnproch551/blob/cf9e2035a1dcc8145db77af51872ac1c767c8823/main.cpp#L439
For some reason there are two `A8` commands sent, with slight differences. Can this second one be commented out as well, or might that fail the `init or erase ???` and brick my device?

DeqingSun, 2023-03-12:
u8FamilyID of ch552 is not 0x12. You can comment it out if you worry about it.

biemster, 2023-03-12:
Ah I see the confusion now! L507 which you initially mentioned is for ID 0x12, but ch552 is apparently not part of that. So the Write on L439 is the one that actually should be commented out.

biemster, 2023-03-12:
So this is the cfg written on every download:
```C
/* Write boot options. On ch552, write 8 bytes from u8WriteBootOptionsCmd[5] to ROM_CFG_ADDR-8 */
/* ch552 only check ROM_CFG_ADDR-4 (written 0x03), bit 1, Set use P3.6 as boot. Clear P1.5. bit 0 related to timeout */
uint8_t u8WriteBootOptionsCmd[64] = {
	0xA8, 0x0E, 0x00, 0x07, 0x00, 0xFF, 0xFF, 0xFF,
	0xFF, 0x03, 0x00, 0x00, 0x00, 0xFF, 0x52, 0x00,
	0x00
};
```
where `0x03, 0x00, 0x00, 0x00, 0xFF, 0x52, 0x00, 0x00` is what ends up between 3FF8H to 3FFFH, and the last 6 bytes is actually the chip ID if I understand the datasheet correctly.
That leaves `0x03, 0x00` which fills the bits outlined in table 6.2 of the CH552DS1.PDF datasheet.
Furthermore, this whole array is present in the binary of `vnproch55x` so I could actually just patch this (instead of recompiling) to set the bootconfig to P1.5.
Looking at the comments in the code, very likely I should change the `0x03` (which has bit 1 and bit 0 set), to something else. But what? (I'll reply back here if/when I figure it out, but pointers are definitely welcome!)

biemster, 2023-03-12:
Confusingly, for the ch551 flasher where this code is based on, this array is called `u8InitCmd` and has the same `0x52` chip ID byte in it (it is exactly the same `A8` command). @DeqingSun am I way off here and just chasing a red herring? Could you explain your comment 
```
ch552 only check ROM_CFG_ADDR-4 (written 0x03), bit 1, Set use P3.6 as boot. Clear P1.5. bit 0 related to timeout
```
a bit more?

DeqingSun, 2023-03-13:
Refer to https://www.mikrocontroller.net/attachment/570882/bootloaderV25.a51

```
    mov   DPTR, #ROM_CFG_ADDR-4  
    clr   A                   ; get the boot config
    movc  A, @A+DPTR
    mov   __cfg, A 
```

And in later code, __cfg determines which pin used to trigger bootloader.

biemster, 2023-03-14:
> Looking at the comments in the code, very likely I should change the `0x03` (which has bit 1 and bit 0 set), to something else. But what? (I'll reply back here if/when I figure it out, but pointers are definitely welcome!)

Changing `0x03` to `0x01` (or anything else) doesn't seem to change the __cfg on the chip at all (which is actually `0x23` when read out). I don't have access to a Windows machine, and the Windows flasher doesn't work under wine anymore, so I don't have another option at the moment then to keep researching what is going on here (I really need bootloader on P1.5 :) )

DeqingSun, 2023-05-12:
the vnproch551 come with commit https://github.com/DeqingSun/vnproch551/commit/721fc2400450e927ca099803def5aa0e731634f4

You can use ```-c KEEP``` to preserve config bytes. And you can add that option to ```tools.vnproch55x_usb.upload.pattern``` in platform.txt to do it automatically.

closed 2023-05-12 by DeqingSun