DeqingSun/ch55xduino #127: Compilation error for board 'CH552 Board'.

ogogon, 2023-03-19:
Greetings!

I connected your board in Arduino IDE and tried to compile an "empty" program (sketch).
```
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

 I didn't try to load anything. Only compiled. Here is what the compiler tells me:
> Arduino: 1.8.19 (Linux), Board:"CH552 Board, 16 MHz (internal), 3.3V or 5V, Default CDC, Serial"
> 
> /snap/arduino/85/arduino-builder -dump-prefs -logger=machine -hardware /snap/arduino/85/hardware -hardware /home/ogogon/snap/arduino/85/.arduino15/packages -tools /snap/arduino/85/tools-builder -tools /snap/arduino/85/hardware/tools/avr -tools /home/ogogon/snap/arduino/85/.arduino15/packages -built-in-libraries /snap/arduino/85/libraries -libraries /home/ogogon/snap/arduino/current/Arduino/libraries -fqbn=CH55xDuino:mcs51:ch552:clock=16internal,usb_settings=usbcdc,upload_method=serial -vid-pid=0000_0000 -ide-version=10819 -build-path /tmp/arduino_build_315454 -warnings=none -build-cache /tmp/arduino_cache_694290 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.sdcc.path=/home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/sdcc/build.13407 -prefs=runtime.tools.sdcc-build.13407.path=/home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/sdcc/build.13407 -prefs=runtime.tools.MCS51Tools.path=/home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/MCS51Tools/2022.10.19 -prefs=runtime.tools.MCS51Tools-2022.10.19.path=/home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/MCS51Tools/2022.10.19 -verbose /tmp/untitled150057972.tmp/sketch_mar20a/sketch_mar20a.ino
> /snap/arduino/85/arduino-builder -compile -logger=machine -hardware /snap/arduino/85/hardware -hardware /home/ogogon/snap/arduino/85/.arduino15/packages -tools /snap/arduino/85/tools-builder -tools /snap/arduino/85/hardware/tools/avr -tools /home/ogogon/snap/arduino/85/.arduino15/packages -built-in-libraries /snap/arduino/85/libraries -libraries /home/ogogon/snap/arduino/current/Arduino/libraries -fqbn=CH55xDuino:mcs51:ch552:clock=16internal,usb_settings=usbcdc,upload_method=serial -vid-pid=0000_0000 -ide-version=10819 -build-path /tmp/arduino_build_315454 -warnings=none -build-cache /tmp/arduino_cache_694290 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.sdcc.path=/home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/sdcc/build.13407 -prefs=runtime.tools.sdcc-build.13407.path=/home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/sdcc/build.13407 -prefs=runtime.tools.MCS51Tools.path=/home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/MCS51Tools/2022.10.19 -prefs=runtime.tools.MCS51Tools-2022.10.19.path=/home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/MCS51Tools/2022.10.19 -verbose /tmp/untitled150057972.tmp/sketch_mar20a/sketch_mar20a.ino
> Using board 'ch552' from platform in folder: /home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/hardware/mcs51/0.0.16
> Using core 'ch55xduino' from platform in folder: /home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/hardware/mcs51/0.0.16
> Detecting libraries used...
> /home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/MCS51Tools/2022.10.19/wrapper/sdcc.sh /home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/sdcc/build.13407/bin/sdcc /tmp/arduino_build_315454/sketch/sketch_mar20a.ino.cpp /dev/null re12 -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mmcs51 -DCH552 -DF_CPU=16000000L -DF_EXT_OSC=0L -DARDUINO=10819 -DARDUINO_ch55x -DARDUINO_ARCH_MCS51 --model-small -DEP0_ADDR=0 -DEP1_ADDR=10 -DEP2_ADDR=20 -I/home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/hardware/mcs51/0.0.16/cores/ch55xduino -I/home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/hardware/mcs51/0.0.16/variants/ch552 -I/home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/sdcc/build.13407/share/sdcc/include
> Generating function prototypes...
> /home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/MCS51Tools/2022.10.19/wrapper/sdcc.sh /home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/sdcc/build.13407/bin/sdcc /tmp/arduino_build_315454/sketch/sketch_mar20a.ino.cpp /tmp/arduino_build_315454/preproc/ctags_target_for_gcc_minus_e.cpp re12 -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mmcs51 -DCH552 -DF_CPU=16000000L -DF_EXT_OSC=0L -DARDUINO=10819 -DARDUINO_ch55x -DARDUINO_ARCH_MCS51 --model-small -DEP0_ADDR=0 -DEP1_ADDR=10 -DEP2_ADDR=20 -I/home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/hardware/mcs51/0.0.16/cores/ch55xduino -I/home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/hardware/mcs51/0.0.16/variants/ch552 -I/home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/sdcc/build.13407/share/sdcc/include
> /snap/arduino/85/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_315454/preproc/ctags_target_for_gcc_minus_e.cpp
> Компиляция скетча...
> /home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/MCS51Tools/2022.10.19/wrapper/sdcc.sh /home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/sdcc/build.13407/bin/sdcc /tmp/arduino_build_315454/sketch/sketch_mar20a.ino.cpp /tmp/arduino_build_315454/sketch/sketch_mar20a.ino.cpp.o re2 -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --less-pedantic -mmcs51 -DCH552 -DF_CPU=16000000L -DF_EXT_OSC=0L -DARDUINO=10819 -DARDUINO_ch55x -DARDUINO_ARCH_MCS51 --model-small -DEP0_ADDR=0 -DEP1_ADDR=10 -DEP2_ADDR=20 -I/home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/hardware/mcs51/0.0.16/cores/ch55xduino -I/home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/hardware/mcs51/0.0.16/variants/ch552 -I/home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/sdcc/build.13407/share/sdcc/include
> /home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/sdcc/build.13407/bin/sdas8051: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/ogogon/snap/arduino/85/.arduino15/packages/CH55xDuino/tools/sdcc/build.13407/bin/sdas8051)
> cp: cannot stat '/tmp/arduino_build_315454/sketch/sketch_mar20a.ino.cpp.o': No such file or directory
> exit status 1
> Compilation error for board CH552 Board.

Please tell me what am I doing wrong?

Ogogon.

DeqingSun, 2023-03-19:
GLIBC_2.29 is the key. Your Linux version is lower than the build machine used by sdcc team or my build machine. So you shall build your own, replace the tool chain and try again

koenvervloesem, 2023-03-21:
I had the same error with the Arduino 1.8.19 snap, probably because it's based on core18 (Ubuntu 18.04, which has glibc 2.27). With the Arduino 2.0 AppImage it works.

closed 2023-05-12 by DeqingSun