DeqingSun/ch55xduino #132: HID keyboard issue

yuan910715, 2023-03-30:
hi 
I use examples : HidKeyboard
when I plug usb cable to my computer and wait about 2 secs and then press the key , it works well. 
but if I press the key immediately after I plug usb cable to my computer , all keys will not available 

When I use the code : CDC and HID together
https://github.com/Noteolvides/CubeTouch/tree/main/Examples/CDC_KEYBOARD
 if I press the key immediately after I plug usb cable to my computer , all keys will not available ,but CDC works well.

I don't know why all keys will not available if I press the key immediately

test1025, 2023-04-21:
我也遇到了这个问题，能不能识别到成了运气问题

DeqingSun, 2023-04-24:
The issue can be reproduced on a Win11 laptop but not on a Mac OS laptop. And this issue will be investigated and tested.

DeqingSun, 2023-04-25:
@yuan910715 @test1025 The updated example should have it fixed.

https://github.com/DeqingSun/ch55xduino/tree/ch55xduino/ch55xduino/ch55x/libraries/Generic_Examples/examples/05.USB/HidKeyboard

yuan910715, 2023-04-25:
Hi  I have tested it and you fixed it , thanks a lot !!

closed 2023-04-25 by yuan910715

yuan910715, 2023-04-25:
@DeqingSun 
By the way, how can I add HID mouse to my project? I used CDC and HID together as a base project, and develop a web page for changing keys map, https://topyuan.top/3key and I want to add a new feature: change one key to mouse left or right,  I can't understand the code in USBconstant and USBhandler ,  can you give me some advices?
https://github.com/yuan910715/3keys/tree/main/ch552/threekey