DeqingSun/ch55xduino #133: Error compiling sample file

smilecat2788, 2023-04-17:
The following content is from machine translation, please forgive me if there is any error
[image]
This example comes from keyboard serial,I haven't made any modifications,This is the complete error message
C:\Users\Song\AppData\Local\Arduino15\packages\CH55xDuino\tools\sdcc\build.11630/bin/sdcc C:\tmp\arduino-sketch-A79BC9F0D66EAA1409A9B3A185A1DEDD\sketch\Serial.ino.cpp nul re12 -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mmcs51 -Dmcs51 -DF_CPU=24000000L -DARDUINO=10607 -DARDUINO_ch55x -DARDUINO_ARCH_MCS51 --model-small -DEP0_ADDR=0 -DEP1_ADDR=10 -DEP2_ADDR=20 -IC:\Users\Song\AppData\Local\Arduino15\packages\CH55xDuino\hardware\mcs51\0.0.7\cores\ch55xduino -IC:\Users\Song\AppData\Local\Arduino15\packages\CH55xDuino\hardware\mcs51\0.0.7\variants\ch552 -IC:\Users\Song\AppData\Local\Arduino15\libraries\Keyboard\src -IC:\Users\Song\AppData\Local\Arduino15\packages\CH55xDuino\tools\sdcc\build.11630/include
Mark re12:C:\Users\Song\AppData\Local\Arduino15\packages\CH55xDuino\tools\sdcc\build.11630/bin/sdcc -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mmcs51 -Dmcs51 -DF_CPU=24000000L -DARDUINO=10607 -DARDUINO_ch55x -DARDUINO_ARCH_MCS51 --model-small -DEP0_ADDR=0 -DEP1_ADDR=10 -DEP2_ADDR=20 -IC:\Users\Song\AppData\Local\Arduino15\packages\CH55xDuino\hardware\mcs51\0.0.7\cores\ch55xduino -IC:\Users\Song\AppData\Local\Arduino15\packages\CH55xDuino\hardware\mcs51\0.0.7\variants\ch552 -IC:\Users\Song\AppData\Local\Arduino15\libraries\Keyboard\src -IC:\Users\Song\AppData\Local\Arduino15\packages\CH55xDuino\tools\sdcc\build.11630/include C:\tmp\arduino-sketch-A79BC9F0D66EAA1409A9B3A185A1DEDD\sketch\Serial.ino.cpp -o nul
cpp found
In file included from C:\TEMP\.arduinoIDE-unsaved2023317-15276-1157qz2.jp8l\Serial\Serial.ino:22:
C:/Users/Song/AppData/Local/Arduino15/libraries/Keyboard/src/Keyboard.h:25:17: fatal error: HID.h: No such file or directory
compilation terminated.

exit status 1

Compilation error: exit status 1

DeqingSun, 2023-04-17:
You can only use the examples come with the CH55xduino package. The Arduino examples may not work.

Use
Generic_Examples/examples/05.USB/HidKeyboard/

smilecat2788, 2023-04-19:
The problem has been resolved,Thank you very much.
[image]
The example here is unusable, but it shows an example of ch552

DeqingSun, 2023-04-23:
There was a bug in IDE2. The examples are actually the blank one.

closed 2023-05-02 by DeqingSun