DeqingSun/ch55xduino #134: How to use Mouse_scroll?

yuan910715, 2023-04-26:
Hi
I use example: USBHIDKeyboardMouse , 
Mouse_click and Mouse_move is Ok ,but Mouse_scroll can't work, how can I use this method? I test Mouse_scroll(0) Mouse_scroll(1) Mouse_scroll(100) Mouse_scroll(200)..

DeqingSun, 2023-04-26:
It seems I forgot to add scroll descriptor. Will fix soon.

DeqingSun, 2023-04-26:
@yuan910715 

https://github.com/DeqingSun/ch55xduino/commit/c5924871c38e2fc7c055c4469927c775a5e6c66e

This commit fixed the bug.

yuan910715, 2023-04-26:
Thanks , I have tested and it's ok
I will update my 3 keys keyboard project soon.

closed 2023-04-26 by yuan910715

yuan910715, 2023-04-26:
Hi, 
I'm in China, do you shop online from China? if you have your transshipment warehouse address in China , please send it to me (admin@toyuan.top) , I will send you a DIY 3 keys keyboard as a gift (https://github.com/yuan910715/3keys), thanks.

DeqingSun, 2023-04-26:
I always wanted to make a marco keyboard and I still have the unassembled parts bought in 2017. I'll email you the address when the PCB factory get my board right. Thank you.

There was also this project
https://twitter.com/ysni_pub/status/1564415437634666496
https://github.com/yswallow/CH552duinoKeyboard
which uses an existing mapping system. I will try to see if I can get it work.

DeqingSun, 2023-04-27:
@yuan910715 
Email was rejected

Address not found
--
Your message wasn't delivered to admin@toyuan.top because the domain toyuan.top couldn't be found. Check for typos or unnecessary spaces and try again.

DeqingSun, 2023-05-26:
@yuan910715 
I received your package yesterday, thanks. 
I make a new firmware on your hardware that is partially compatible with the [QMK Firmware](https://qmk.fm/). All interfaces are HID so there is no need to install any driver. And the keyboard can be configured with [via app](https://usevia.app/). If the dataflash is cleared. The firmware will automatically restore it to copy/paste/tab, the most common keys I use. And the keyboard will use the correct modifier of control or command, depending on you host OS.