DeqingSun/ch55xduino #136: CH552 strange behavior with USB port

XanCraft21, 2023-05-12:
Hello, i have discovered that my controller is behaving very strangely. All of a sudden i can upload through the usb port without a FTDI adapter. Before i needed the adapter to even program the board, but the USB upload never worked before.

Even though i get a port that is only for Serial printing, it still wouldn’t upload. But the strangest thing happened.

I enabled the user code in the USB setting to test one of the examples, then after that i thought i would try uploading it through USB again then it just suddenly started working correctly. It still continued ti work even after setting the USB mode back to default CDC.

I believe it is a problem with my board, and it confuses me a lot. If anyone knows why it all of a sudden started working as intended please tell me. This board has been nothing but trouble since i got it.

usbman01, 2023-05-15:
The bootloader supports USB and serial download. When the loader is activated and there is activity on Rx the loader assums serial mode and stays there. Without activity on Rx USB mode is activated.

The loader can be activated by HW contition on D+ (default) or P1.5 (since v2.31 if programmed) during Power On Reset. (POR only)
I foud that activating the loader in v2.5 is not that stable as before.
Its also possible to activate the loader by jmp 0x3800. This software activation does not work on recent controllers like ch549. Software activation  limits some settings during flashing. Its for example not possible to change the bootpin.

XanCraft21, 2023-05-15:
Hmm, that still doesn’t explain why my board just suddenly started working as it should have, but the info helps a little.

The USB upload on my board has never worked until i set the USB mode to User Code with Ram enabled. It still worked after setting back to default CDC mode.

It may be because the company i got it from did a poor job at the manufacturing and may have used a slightly defective chip, or the chip manufacturer WCH did a poor job and the pcb manufacturer probably made it worse.

The company is WeActStudio and i bought some products from them on AliExpress and they turned out to be not that good. I could complain about it all day but i won’t do it here.

XanCraft21, 2023-05-16:
Hmm that could be why i was having that problem. I had the latest version of the software package from 0.0.17 but i had to re-download it because there was a bug fix for a previous issue i posted.

i only have one board and i might get another one to see if the problem is the same there. Thank you for your help.

closed 2023-06-02 by DeqingSun