DeqingSun/ch55xduino #137: Is it possible to use sqrt() in ch552?

ZerooNight, 2023-05-29:
[image]
i try sqrt(2),and it shows the error

DeqingSun, 2023-06-29:
The compiler SDCC does not have sqrt function but has sqrtf. Unfortunately an earlier fix to add --int-long-reent for https://github.com/DeqingSun/ch55xduino/issues/107 did not deal with library properly. So there will be an link error. 

I'll fix the issue later.

DeqingSun, 2023-07-03:
Please try sqrtf in 0.0.18

ZerooNight, 2023-07-04:
HI
thanks for the update
but it seems that any code are not working in 0.0.18
the comport doesn't show after burn

take a look at the pic below
i wrote a simple test
after update to 0.0.18,the usage of program space reduce a lot
[image]
[image]

DeqingSun, 2023-07-04:
Sorry I just realized there was a compiler problem in Windows. If you have a Mac or Linux it would work. I'm still investigating the reason

DeqingSun, 2023-07-05:
The 0.0.18 has been rolled backed to an earlier compiler built with updated library. Please remove the package and add again.

ZerooNight, 2023-07-05:
[image]
works fine now!
thank you very much

closed 2023-07-05 by DeqingSun