DeqingSun/ch55xduino #140: Low Speed USB - Device Descriptor Fails

Saatvik-Aggarwal, 2023-06-06:
When editing the `USBDeviceCfg` in USBhandler.c to use Low Speed USB, the communication does not work. I've been trying to add support for lower clock frequencies down to 1.5MHz so others can save power, but any frequencies under 6MHz require low speed USB, which does not work. I've tested on the CH552G.

Here is my version

```c++
void USBDeviceCfg()
{
    USB_CTRL = 0x00;                                                           //Clear USB control register
    USB_CTRL &= ~bUC_HOST_MODE;                                                //This bit is the device selection mode
    USB_CTRL |=  bUC_DEV_PU_EN | bUC_INT_BUSY | bUC_DMA_EN;                    //USB device and internal pull-up enable, automatically return to NAK before interrupt flag is cleared during interrupt
    USB_DEV_AD = 0x00;                                                         //Device address initialization
         USB_CTRL |= bUC_LOW_SPEED;
         UDEV_CTRL |= bUD_LOW_SPEED;                                                //Run for 1.5M
    //USB_CTRL &= ~bUC_LOW_SPEED;
    //UDEV_CTRL &= ~bUD_LOW_SPEED;                                             //Select full speed 12M mode, default mode

#if defined(CH551) || defined(CH552) || defined(CH549)
    UDEV_CTRL |= bUD_PD_DIS;                                                     // Disable DP/DM pull-down resistor
#endif
#if defined(CH559)
    UDEV_CTRL = bUD_DP_PD_DIS;                                                     // Disable DP/DM pull-down resistor
#endif
    UDEV_CTRL |= bUD_PORT_EN;                                                  //Enable physical port
}
```

I get this error: "String Descriptor for index 2 not available while device is in low power state."
If I remove the string descriptors, I get device descriptor request failed.

DeqingSun, 2023-06-06:
I haven't used low speed for a long time. I don't remember clearly but you may need to change descriptor from USB 2.0 to 1.1 or 1.0.

usbman01, 2023-06-17:
try to change DEFAULT_ENDP0_SIZE. If i remember correctly only 0x08 is allowed for low speed.

DeqingSun, 2024-03-29:
Close for inactivity.

closed 2024-03-29 by DeqingSun