DeqingSun/ch55xduino #147: CH554 support

mcu23, 2023-10-09:
Hello,  At v 0.0.17  code upload possible to CH554. But  not available at v0.0.18 and  v0.0.19 

I know this problem related  code size and  MCU ID ( 54 11) mismatch while programming.  

Would you consider newer version for solution? 

Thanks

DeqingSun, 2023-10-09:
@mcu23 Thanks for reporting. It looks weird as the source code treated CH554 the same way as CH552, can you paste the whole error message? Especially I want to see "Updated on" section.

mcu23, 2023-10-09:
`
.....\AppData\Local\Temp\arduino\sketches\06525571E1802E23C9E6813E53E3E029/BlinkCH55xDuino.ino.hex"
------------------------------------------------------------------
CH55x Programmer by Deqing
Updated on: 2023/06/01
------------------------------------------------------------------
usbRertySeconds 2
target: CH552
config bytes: 3
Load file as hex
   Loaded 4674 bytes between: 0000 to 128F
ch375Version 35
CH375GetUsbID 55e04348
CH375 open OK
MCU ID: 54 11
Found Device CH554
Target in argument CH552 doesn't match detected target CH554
Failed uploading: uploading error: exit status 1`

mcu23, 2023-10-09:
[error1.txt](https://github.com/DeqingSun/ch55xduino/files/12847848/error1.txt)

DeqingSun, 2023-10-10:
@mcu23 Please remove your CH55xduino package (should be able to do in board manager) and add 0.0.19 again.

The new version of upload tool should work with "Updated on: 2023/10/09"

DeqingSun, 2023-10-10:
I have a feeling the last command of uploader tool to restart the chip did not get a proper response. The response may be lost in CH552 or windows side. But as long as you see "Verify complete!!!", the firmware is ready.

Does the LED blink immediately after the fail message? If so the CH552 got the reboot command but the tool does not get respond. I'll do some test on my side. Which version of windows are you using?

mcu23, 2023-10-10:
Yes LED blink immediately after the fail message. Windows 7 used.

DeqingSun, 2023-10-10:
@mcu23 DIY board may be a possible. Can you show you schematic and board picture? I'll also look for a win7 computer and v2.4.0 bootloader chip. If you have a logic analyzer or USB analyzer you can also try to monitor the communication at the end of upload process.

mcu23, 2023-10-10:
I think, USB module ON by default.  But no usb stack  in blink hex?
So Windows can effort for driver etc.

DeqingSun, 2023-10-10:
Please try this upload tool. You can use this one to replace the old one come in the package as I found one code that might cause the issue.

You should see "Updated on: 2023/10/10"

[win_build.zip](https://github.com/DeqingSun/ch55xduino/files/12858248/win_build.zip)

mcu23, 2023-10-10:
Thank you for new build. This content placed  to 
..Myusername\AppData\Local\Arduino15\packages\CH55xDuino\tools\MCS51Tools\2023.10.09\win
(busybox.exe not changed)

Same result. I think,firmware  realy contains CDC stack
USBlyzer Report:
[cdcserial.pdf](https://github.com/DeqingSun/ch55xduino/files/12858482/cdcserial.pdf)

Post programming info:
[Error_10_10_23_2.txt](https://github.com/DeqingSun/ch55xduino/files/12858489/Error_10_10_23_2.txt)

DeqingSun, 2023-10-10:
It seems the ch552 did receive the reboot signal but the computer did not even think the ch552 get it.

I'll try to see if I can reproduce the error.

closed 2024-03-30 by DeqingSun