DeqingSun/ch55xduino #149: Idea: add TinyRF capability to open the ch55x to 315/433MHz communication

stripu, 2023-10-26:
Porting the TinyRF library (https://github.com/pouriap/TinyRF) would really open a lot of possibilities, as it would enable usb port communications to low power RF devices.
The library is not too big and it's optimized for low memory devices, so I believe it might suit the ch55x family.
Unfortunately I'm not capable of porting something like this, but hopefully someone might :)

Thanks!

DeqingSun, 2023-10-26:
Looks interesting. I'll look for the leftover $0.3 433M modules to get started. 
How is the library compare to the radio head one? https://www.airspayce.com/mikem/arduino/RadioHead/

DeqingSun, 2023-10-26:
Some quick compare:

https://github.com/pouriap/TinyRF/blob/master/src/TinyRF_TX.cpp

https://github.com/sparkfun/SparkFun_RadioHead_Arduino_Library/blob/master/src/RH_ASK.cpp

TinyRF uses pulse-width encoding (No consecutive 0 or 1 guaranteed) and Radiohead uses high low directly with 4B6B encoding. 

Their preambles are a bit different

stripu, 2023-10-27:
Indeed, the two libraries are quite different in how they do things.
Nevertheless, I've used them both and to be honest I prefer the TinyRF one in terms of reliability and ease of use :)
I'm particularly interested in the RX part, in order to use the ch55x board connected to USB and simulate different button presses when a RF signal is received.

Here's hoping that you are successful in your tests, this would be an amazing addition!

closed 2026-01-12 by DeqingSun