DeqingSun/ch55xduino #150: Keyboard with boot protocol enabled

OmerFS, 2023-10-26:
Hello,

ch55xduino/ch55x/libraries/Generic_Examples/examples/05.USB/HidKeyboard
compiled to ch552 mcu HID working fine on windows and linux. it is not working on bios

How to made work on bios screen? 

(by the way, it is working on new model computers bioses but not working on a bit old bioses like Asus N550JV)

Thanks in advance

DeqingSun, 2023-11-11:
I can confirm the emulated keyboard does not work on the boot screen of a Mid-2014 macbook pro while a real Dell keyboard does work. 
So there must be some implementation problem in the boot protocol. I will take a look soon.

DeqingSun, 2023-11-12:
For progress record:

There is some progress in https://github.com/DeqingSun/ch55xduino/blob/ec3b8958397dac77035ab4ae40cd06331b51320d/ch55xduino/ch55x/libraries/Generic_Examples/examples/05.USB/HidKeyboard/src/userUsbHidKeyboard/USBhandler.c that deal with HID report access on EP0. More work need to be done to do get EP0 IN work properly and send report over EP0 instead of EP1 when keyboardProtocol is set to 0.

DeqingSun, 2023-11-14:
@OmerFS I seems get it working on my Mac in boot screen. Let me know if this example work for you.

https://github.com/DeqingSun/ch55xduino/tree/19f3f8c1b5a86ab0c537f0cf059bbe71e6849bd1/ch55xduino/ch55x/libraries/Generic_Examples/examples/05.USB/HidKeyboard

OmerFS, 2024-03-30:
Hello i am sorry for not returning in time i saw your post in my mail today i have variety of old notebooks i will test and post the results thank you so much.

DeqingSun, 2024-04-03:
Thanks for sharing. I suppose there must be some USB request that I did not handle properly. 

I'll check it this weekend with a real windows machine. I suppose the bios in windows and mac still handle the USB differently.

DeqingSun, 2024-04-04:
I've tested it on 
Lenovo Thinkpad t410 2518-f4u 2010
Acer Aspire 4930 JAL90 2008
They both work in Bios.
I'll take a look at the other projects.

closed 2025-06-08 by DeqingSun