DeqingSun/ch55xduino #151: About absolute mouse？鼠标绝对坐标移动问题求助

wukongxuetang, 2023-11-01:
测试HID Keyboard mouse combo example,鼠标指针只能相对移动，我知道arduino有一个Absmouse库，可以设置鼠标的绝对坐标，不知道ch55x能不能通过修改hid描述符的方式实现？

Testing the HID Keyboard mouse combo example, the mouse pointer can only move relative. I know that Arduino has a library named Absmouse that can set the absolute coordinates of the mouse, but I'm not sure if it can be achieved by modifying the hid descriptor？

Thanks！

wukongxuetang, 2023-11-01:
just this is a sample for absolute mouse
https://github.com/earlephilhower/arduino-pico/blob/master/cores/rp2040/sdkoverride/tusb_absmouse.h

DeqingSun, 2023-11-01:
Yes,
https://github.com/DeqingSun/ch55xduino/blob/1f7bfe1739d98d2beebc8d6e62aaa7aec69d2076/ch55xduino/ch55x/libraries/Generic_Examples/examples/05.USB/HidKeyboardMouse/src/userUsbHidKeyboardMouse/USBconstant.c#L145
Change it to something like:
https://github.com/earlephilhower/arduino-pico/blob/319d36531ed63aa13f97544f18e6de010a0d5e46/cores/rp2040/sdkoverride/tusb_absmouse.h#L63

wukongxuetang, 2023-11-02:
https://github.com/windoze/esparrier-idf/blob/c056fbfd47dababbdad1c426ace1023cfc50045f/usb_util/usb_util.c#L17

谢谢您的指点，不过对于硬件底层确实是一窍不通。
Sorry, I can only be considered as an application developer and am not very familiar with hardware. I am not familiar with hid descriptors and cannot correspond to macro definitions and numerical encoding. I hope to receive more help from you.Thanks

wukongxuetang, 2023-11-03:
I found this code,hope it can work

static const uint8_t HID_REPORT_DESCRIPTOR5[] PROGMEM = {
	0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
	0x09, 0x02,        // Usage (Mouse)
	0xA1, 0x01,        // Collection (Application)
	0x09, 0x01,        //   Usage (Pointer)
	0xA1, 0x00,        //   Collection (Physical)
	0x85, 0x01,        //     Report ID (1)
	0x05, 0x09,        //     Usage Page (Button)
	0x19, 0x01,        //     Usage Minimum (0x01)
	0x29, 0x05,        //     Usage Maximum (0x05)
	0x15, 0x00,        //     Logical Minimum (0)
	0x25, 0x01,        //     Logical Maximum (1)
	0x95, 0x05,        //     Report Count (5)
	0x75, 0x01,        //     Report Size (1)
	0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
	0x95, 0x01,        //     Report Count (1)
	0x75, 0x03,        //     Report Size (3)
	0x81, 0x03,        //     Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
	0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
	0x09, 0x30,        //     Usage (X)
	0x09, 0x31,        //     Usage (Y)
	0x16, 0x00, 0x00,  //     Logical Minimum (0)
	0x26, 0xFF, 0x7F,  //     Logical Maximum (32767)
	0x36, 0x00, 0x00,  //     Physical Minimum (0)
	0x46, 0xFF, 0x7F,  //     Physical Maximum (32767)
	0x75, 0x10,        //     Report Size (16)
	0x95, 0x02,        //     Report Count (2)
	0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
	0xC0,              //   End Collection
	0xC0               // End Collection
};

closed 2024-03-29 by DeqingSun