DeqingSun/ch55xduino #152: Drivers problem

veso266, 2023-11-08:
Hello there, I got one of CH552 and for the love of me I cannot seam to program in
I can enter bootloader mode (by pressing PJ5 and plugging in USB cable, the device will enter bootloader, stay there for 9seconds then exit to normal mode

But the problem is that USB VID and PID are 0
[image]

I cannot install any driver on my windows machine (tried to install WinUSB with zadig), but it says the some force mode is required
```
Using inf name: Unknown_Device.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\Mitja\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\Mitja\usb_driver\Unknown_Device.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:info [wdi_prepare_driver] Test signing is: Disabled
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Mitja\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Mitja\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'unknown_device.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Mitja\usb_driver\driver\unknown_device.inf'
libwdi:debug [AddFileHash] 'unknown_device.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Mitja\usb_driver\unknown_device.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Mitja\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Mitja\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\Mitja\usb_driver\Unknown_Device.cat'
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_0000&PID_0000 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_0000&PID_0000 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\VID_0000&PID_0000 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\VID_0000&PID_0000 (libwdi autogenerated)'
libwdi:info [SelfSignFile] Added certificate 'CN=USB\VID_0000&PID_0000 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\Mitja\usb_driver\Unknown_Device.cat'
libwdi:info [SelfSignFile] Successfully deleted private key
libwdi:debug [wdi_install_driver] Using progress bar mode
libwdi:debug [installer process] got parameter Unknown_Device.inf
libwdi:debug [process_message] Got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_0000&PID_0000\6&46B13AD&0&3'
libwdi:debug [process_message] Got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\UNKNOWN'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-68339179-3683065201-3801443906-1000'
libwdi:debug [installer process] using syslog 'C:\Windows\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] Switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\UNKNOWN - please wait...
libwdi:debug [process_message] Switching timeout back to finite
libwdi:debug [installer process] A more recent driver was found (force option required).
libwdi:debug [process_message] Installer process completed
Driver Installation: FAILED (Resource already exists)
```

and because I have no driver, there is no way to install anything using Arduino IDE

Why are my VID and PID 0 (is this normal?)
Can I somehow change them (it doesn't matter to what, just something else then 0 so I can install the driver)

I mean my bootloader doesn't seam to be coruupted, because I can enter it and it will exit after 9 seconds, its just that USB VID and PID are 0

DeqingSun, 2023-11-08:
It feels like a hardware issue. Did you made your own board? Did you have capacitors on 3.3V and 5V?
Did you try another computer.

veso266, 2023-11-09:
Hmm a hardware issue
I did not made my own board, I found the board in the dumbster (maybe thats the reason why it was there in the first place)

I did not have any capacitors, just did like this guy did (have the same board): https://www.youtube.com/watch?v=Y5rnfOp5gMM&t=201s
Holding P36 while plugging USB (or alternative way, hold P36, click the RST key, and release the P36 key 0.5S later) to enter bootloader mode, which it secsesfully does, its just that VID and PID are 0 for some strange reason

If its a hardware issue, do you maybe have any idea what could be wrong (cold solder joint maybe)
This board is so small that I bearly see anything on it, but I would still be happy if I could fix a thing someone thought its for disposal (the earth is more sustainable that way)

DeqingSun, 2023-11-09:
Try another computer, preferable linux/mac to avoid any driver hassle. If it still doesn't work, it does not worth fixing. Trouble shooting USB issue will need a lot of equipment and get a new board actually generate less waste.

closed 2023-11-10 by DeqingSun