DeqingSun/ch55xduino #153: TouchKey_begin disrupts functionality of pins not enabled for touch

buttim, 2023-11-26:
I am using pins P1.0 and P1.1 for soft I2C in my code. As soon as I enabled touch sensing for pin 1.5 the I2C interface stopped working.
I thoroughly examined the touch implementation and found that in many parts of the code all 6 touch channels are activated, even if not enabled. This stops the pins not enabled for touch from functioning normally as GPIO.
I investigated the issue and found that adding a few lines of code seems to correct the problem.
All changes affect file `TouchKey.c`
In function `TouchKey_ISR_Handler`, from line  45, I changed

```
  index++;
  if (index > 6)
    index = 1;
```
    
to

```
    do {
      index++;
      if (index>6) index = 1;
    } while ((channelEnabled & (1<<index))==0);
```

In function `TouchKey_begin`, from line 96, I changed:

```
  for (uint8_t i = 0; i < 6; i++) {
    TKEY_CTRL = bTKC_2MS | (1 + i);
    while ((TKEY_CTRL & bTKC_IF) == 0)
      ;
    touchBaseline[i] = TKEY_DAT;
  }

  TKEY_CTRL = bTKC_2MS | 1;
```

to

```
    for (uint8_t i=0;i<6;i++){
      if (channelEnabled & (1<<i)){
        TKEY_CTRL = bTKC_2MS | (1+i);
        while((TKEY_CTRL&bTKC_IF) == 0);
        touchBaseline[i]=TKEY_DAT;
      }
    }
    
    for (uint8_t i=0;i<6;i++){
      if (channelEnabled & (1<<i)) {
	TKEY_CTRL = bTKC_2MS | (1+i);
	break;
      }
    }
```

DeqingSun, 2023-11-27:
Thanks for the contribution. I'll check it throughly later.

DeqingSun, 2024-03-29:
Thank you, the code looks good.

It has been added as https://github.com/DeqingSun/ch55xduino/commit/e66181f61a141df41fc80a3036040365ab39fa3f

closed 2024-03-29 by DeqingSun