DeqingSun/ch55xduino #155: [Demo] Ch554 USB Audio Speaker

ziv2013, 2024-01-05:
Hi,

Thanks for you great work on Ch55x with Arduino. It's quite easy to use. 

I have made a USB Audio Speaker based on your USB MSD demo.

You can find my project at 

https://mc.dfrobot.com.cn/thread-317671-1-1.html#lastpost

And you can have a try if you have a speaker.

Z.t

DeqingSun, 2024-03-04:
Thanks, that opens a lot possibilities. I will check how it works.

ziv2013, 2024-03-05:
> Thanks, that opens a lot possibilities. I will check how it works.

Yes, it would be intresting. But I have found a limitation that the max USB buffer size of Ch55x is 128bytes, which is smaller than 192 bytes. Data lost would happen in every package.

closed 2024-03-05 by ziv2013

DeqingSun, 2024-03-29:
I tried to reduce the sample rate to 24K and use mono sound, now the data packet is 48 bytes.
Also I used timer interrupt to update PWM, and there is walk-around for PWM bug.

The sound is actually pretty good.

https://github.com/DeqingSun/ch55xduino/tree/9c81b75db732fda867513001db168247cd4bf64d/ch55xduino/ch55x/libraries/Generic_Examples/examples/05.USB/SoundCard

ziv2013, 2024-03-29:
> I tried to reduce the sample rate to 24K and use mono sound, now the data packet is 48 bytes. Also I used timer interrupt to update PWM, and there is walk-around for PWM bug.
> 
> The sound is actually pretty good.
> 
> @ziv2013 here is the code
> 
> https://github.com/DeqingSun/ch55xduino/tree/9c81b75db732fda867513001db168247cd4bf64d/ch55xduino/ch55x/libraries/Generic_Examples/examples/05.USB/SoundCard

That's great. I will try your code later.

I'm studying the Ch32V305 these days which USB buffer is big enough. And it's support USB High Speed and I2S.

ziv2013, 2024-04-01:
> I tried to reduce the sample rate to 24K and use mono sound, now the data packet is 48 bytes. Also I used timer interrupt to update PWM, and there is walk-around for PWM bug.
>
> The sound is actually [...]

It works well. I have tested yesterday.