DeqingSun/ch55xduino #158: CH546L unresponsive after flash

sinitax, 2024-02-26:
After uploading a blank sketch to a CH546L using version 0.20.0, the device is not recognized via USB anymore and will not enter DFU mode (does not show as 4348:55e0 or any usb device after pulling P1.5 down during PoR).

The following settings were used in the Arduino IDE:
- USB Settings: Default CDC
- Upload Method: USB
- Clock Source: 24 Mhz (internal)
- Bootloader pin: P1.5 pull-down

According to the manufacturer "In terms of software and hardware, CH549 is compatible with CH546" (https://www.wch-ic.com/products/CH546.html).

I attempted flashing via serial using TXD1 & RXD1, however the programmer did not show any sign of life.

Any idea what the cause is?

sinitax, 2024-02-26:
Great to hear you are looking into it!

The WCHISPTool programmer mentions P1.5 and P5.1. Pulling these up or down through a 10k resistor had not effect (no usb communication).

I took another look at the datasheets for [CH546/547](https://www.wch-ic.com/downloads/CH547DS1_PDF.html) & [CH548/589](https://www.wch-ic.com/downloads/CH549DS1_PDF.html) and it seems like in terms of hardware CH548/CH549 and CH546/CH547 are only pair-wise compatible (e.g. CH546 and CH548 but not CH546 and CH549). The incompatibility stems from varying SFR placement (Section 5.2) and shifted memory map due to smaller total flash size (32kB vs 64kB) (Section 6.1), the latter causing ROM_CFG_ADDR to end up at 0x7FFE instead of 0xFFFE for the CH549 I assume.

[image]

I suspect this caused the config information to be overwritten as part of the application code - but i am not familiar with the programming process in detail.

The datasheet also mentions "Without Bootloader, application code and Bootloader may only download through specialized programmer". I found out that "CH546 [...] only provides UART0" but even this port is unresponsive. Are you familiar with the aforementioned programmer - or is this the serial programming protocol implemented by vnproch55x?

DeqingSun, 2024-02-27:
As far as I know. Wch is not willing to provide serial spi programming to retail user.

sinitax, 2024-04-14:
Hello,

have you found time to test with the CH546G? I got my hands on a few more CH546L and decided to try my luck by creating a new board target and modifying just `upload.maximum_size` to 28672 (0x7000). Upon successful upload, the chip registers again as `Generic CH55xduino`, but is inaccessible again after power cycle (same behavior as before). I tried once with and once without adjusting `DATA_FLASH_ADDR`, `BOOT_LOAD_ADDR` or `ROM_CFG_ADDR` for 32k flash (-0x8000).

Would you have adjusted the addresses differently? - or have an idea what the problem is?

DeqingSun, 2024-04-14:
Yes I did get a few CH546G. I will try to see the communication between the official tool and the chip these days.

closed 2025-01-12 by sinitax