DeqingSun/ch55xduino #160: CH552 into sleep mode and wake it up or switch to low power

HYDRA0422, 2024-04-16:
I am a beginner, and I am looking for examples of how to put the CH552 into sleep mode and wake it up or switch to low power mode for a project powered by a lithium battery. Could you please tell me if the current examples include any such code? Thank you very much！

DeqingSun, 2024-04-16:
No the current example does not include any code with sleep. You need to check the datasheet.

closed 2024-04-16 by DeqingSun

rew1ndustry, 2025-04-24:
so far i have tried this, in setup()

  // preparing for sleep
  ES = 1;  // enable serial interrupts
  EA = 1;  // Global interrupt enable
  SAFE_MOD = 0x55; SAFE_MOD = 0xAA; WAKE_CTRL |= bWAK_BY_USB;
  SAFE_MOD = 0x55; SAFE_MOD = 0xAA; WAKE_CTRL |= bWAK_RXD0_LO;
  SAFE_MOD = 0x55; SAFE_MOD = 0xAA; PIN_FUNC &= ~bUART0_PIN_X;
  // there are other WAKE_CTRL as well, see documentation.

and then just call

  PCON |= 0x01;  // Enter IDLE mode

or

  PCON |= 0x02;  // Enter DEEP SLEEP mode

in my experiments it seems to go to sleep ok, and it seems to wake on USB activity, but so far i cannot get it to wake on incoming serial, as i had hoped.