DeqingSun/ch55xduino #161: Softreset after download CH549

usbman01, 2024-04-25:
I am testing some code on the ch549 using usbcdc. So far  all is working. I can see that after download the soft reset is working. The problem is after the soft reset the usb module is still active and CDC can't be started.

So reset after download does not make any sense. I still need to reboot to activate CDC.

Is it possible to download without that reset?

DeqingSun, 2024-04-25:
Did you try with the official WCHISPtool to upload and reset? If it works then there is bug in my code.
If it does not, then there is nothing I can do.

usbman01, 2024-04-25:
behavour is the same tested with wchisptool v3.2. It seems there is no unplug event or the event is too short to be recognized by win. So I guess i have to live with that. 

I am still trying to hack this thing but up to now without any success. Even that readable part of the bootloader is writeprotected in user mode.

DeqingSun, 2024-04-25:
Do you want a tool to solve this kind of issue?

[[usbman01](https://github.com/usbman01)](https://github.com/DeqingSun/ch55xduino/tree/ch55xduino/pcb/ch55xRebootTool)

I can mail you one assembled one from my place or mail you parts for 5 sets if you don't mind do some soldering.

usbman01, 2024-04-25:
many thanks for your offer but i do not need that. For the power cycle i am using a hub with buttons for Vusb and all of my boards have a jumper for boot.. Thats good enough for me.

closed 2024-04-26 by usbman01