DeqingSun/ch55xduino #162: CH558L unresponsive after first flash

dirkx, 2024-05-16:
Although the datasheet suggests the right boot pin (P5.1/DP), etc -- the CH558L does not seem to respond after its initial (successful) flash.

```
dch559 % ..../vnproch55x -r 2 -t CH558 -c 1 Blink.ino.hex 
------------------------------------------------------------------
CH55x Programmer by Deqing
Updated on: 2023/10/10
------------------------------------------------------------------
usbRertySeconds 2
target: CH558
config bytes: 1
Load file as hex
   Loaded 4919 bytes between: 0000 to 1382
DeviceVersion of CH55x: 2.40 
MCU ID: 58 11
Found Device CH558
Bootloader: 2.4.0
ID: 20 3C 4E BC
XOR Mask: 66 66 66 66 66 66 66 BE 
Write 4995 bytes from bin file.
..................................................
Write complete!!!
Verify chip
..................................................
Verify complete!!!
------------------------------------------------------------------
Reset OK
dch559 % ..../vnproch55x -r 2 -t CH558 -c 1 Blink.ino.hex 
------------------------------------------------------------------
CH55x Programmer by Deqing
Updated on: 2023/10/10
------------------------------------------------------------------
usbRertySeconds 2
target: CH559
config bytes: 1
Load file as hex
   Loaded 4962 bytes between: 0000 to 13AE
No CH55x USB Found, retry...
...
Found no CH55x USB
Time limit reached, exit process
```

it does appear fine on the USB bus though:

```
Bus 004 Device 023: ID 1209:c550 Generic 
```

Does this ring a bell with anyone ?

DeqingSun, 2024-05-16:
The upload tool will change setting to a CH559, but not on CH558

https://github.com/DeqingSun/vnproch551/blob/bd6a1a85be100bda11a7ddc3db55691bf40468cf/main.cpp#L503

So your CH558L kept the factory settings. And you can try another boot pin P4.6 down.

Also the 1209:c550 means the USB stack for CDC is working. You can just set serial port to it and it can automatically go to bootloader in upload process.

dirkx, 2024-05-16:
Ah - understood. And indeed that works.  Will see if I can modify main.cpp/503 to also work for the CH558L.

The USB-CDC stack for the default Blinken appears as /dev/ttyAMC0 (1209:c550) as opposed to the 4348:55e0 when it is powered up with boot pressed (in my case P5.1 to 3V3). And this /dev/ttyAMC0 (1209:c550) behaves as a serial port.

Can this serial port be brought into boot loader mode without a powercycle ? i.e. programatically ?

closed 2024-05-16 by dirkx

DeqingSun, 2024-05-16:
@dirkx Yes, it can be done in default Arduino way, open and close port at 1200bps. And if you choose that port in Arduino IDE and press Upload, it will be done automatically.
Code is located at https://github.com/DeqingSun/ch55xduino/blob/d69a0e446aa082b35a1be808a8c7e22a4886b6e2/ch55xduino/ch55x/cores/ch55xduino/USBCDC.c#L61

DeqingSun, 2024-05-16:
@dirkx I also tried to modify the code with a text editor and Github action compiled the binary. The code is not tested on hardware, but it will likely solve your CH558 problem.

commit:
https://github.com/DeqingSun/vnproch551/commit/c07331616c9802a1ca1510dde47b5e09261b8660

Binary:
[linux_arm64_build.zip](https://github.com/DeqingSun/ch55xduino/files/15340772/linux_arm64_build.zip)
[win_build.zip](https://github.com/DeqingSun/ch55xduino/files/15340773/win_build.zip)
[mac_build.zip](https://github.com/DeqingSun/ch55xduino/files/15340775/mac_build.zip)
[linux_build.zip](https://github.com/DeqingSun/ch55xduino/files/15340776/linux_build.zip)

dirkx, 2024-05-17:
> @dirkx Yes, it can be done in default Arduino way, open and close port at 1200bps. And if you choose that port in Arduino IDE and press Upload, it will be done automatically. Code is located at [...]

Thanks - for splendidly (will try to make a pull request for the right menu/config settings).

dirkx, 2024-05-17:
Added PR #163 with the changes that made it all work automatically from within the Arduino environment (as opposed to tweaking things from the command line).