DeqingSun/ch55xduino #164: WS2812 Brightness

iharshadec, 2024-05-18:
Hi,
I am using WS2812 Chase example which is coming in example. I see that WS2812 is turning ON with very low brightness.
Is there any way we can increase the brightness?

DeqingSun, 2024-05-18:
Have you ever read the code before you ask this question?

iharshadec, 2024-05-19:
> Have you ever read the code before you ask this question?

Yes, In this example code didn't find brightness.
```
#include <WS2812.h>

#define NUM_LEDS 8
#define COLOR_PER_LEDS 3
#define NUM_BYTES (NUM_LEDS*COLOR_PER_LEDS)

#if NUM_BYTES > 255
#error "NUM_BYTES can not be larger than 255."
#endif

__xdata uint8_t ledData[NUM_BYTES];

void setup() {
  pinMode(15, OUTPUT); //Possible to use other pins. 
}

void loop() {

  for (uint8_t i = 0; i < NUM_LEDS; i++) {
    set_pixel_for_GRB_LED(ledData, i, 1, 0, 0); //Choose the color order depending on the LED you use. 
    neopixel_show_P1_5(ledData, NUM_BYTES); //Possible to use other pins. 
    delay(100);
  }
  for (uint8_t i = 0; i < NUM_LEDS; i++) {
    set_pixel_for_GRB_LED(ledData, i, 0, 1, 0);
    neopixel_show_P1_5(ledData, NUM_BYTES);
    delay(100);
  }
  for (uint8_t i = 0; i < NUM_LEDS; i++) {
    set_pixel_for_GRB_LED(ledData, i, 0, 0, 1);
    neopixel_show_P1_5(ledData, NUM_BYTES);
    delay(100);
  }

}

```

DeqingSun, 2024-05-19:
Have you ever tried to explain yourself what does each function and each parameter do? Did you check their prototypes and code?

closed 2024-05-21 by iharshadec