DeqingSun/ch55xduino #167: keyboard_getLEDStatus() does not respond to a NUM LOCK, CAPS LOCK or SCROLL LOCK change

Archetix, 2024-05-24:
Hello.
I have a problem with the Keyboard_getLEDStatus() function.
It seems that its output value does not change depending on the keyboard status leds.

I have tried it on windows 7, windows 10 and also on Ubuntu.

I tried Arduino IDE 1.8.19 and Arduino IDE 2.3.2 but the result is the same.

My test code:

// Microcontroller: CH552G
// Clock Source 24MHz (internal)
// USB Settings: "USER CODE w/148B USB ram"

#ifndef USER_USB_RAM
#error "This example needs to be compiled with a USER USB setting"
#endif

#include "src/userUsbHidKeyboard/USBHIDKeyboard.h"
#define BUTTON1_PIN 15
bool button1PressPrev = false;
uint8_t status;

void setup() {
  USBInit();
  pinMode(BUTTON1_PIN, INPUT_PULLUP);
  status = Keyboard_getLEDStatus();  //initialization state
}

void loop() {

  if (status != Keyboard_getLEDStatus()) { //when anything changes in Keyboard_getLEDStatus()
    Keyboard_write('C');
    Keyboard_write('h');
    Keyboard_write('a');
    Keyboard_write('n');
    Keyboard_write('g');
    Keyboard_write('e');
    delay(500);
    status = Keyboard_getLEDStatus();
  }

  bool button1Press = !digitalRead(BUTTON1_PIN); //manual test by button
  if (button1PressPrev != button1Press) {
    button1PressPrev = button1Press;
    if (button1Press) {
      status = 0;
    }
  }

}

when I press NUM LOCK, CAPS LOCK or SCROLL LOCK on the computer keyboard, the function keyboard_getLEDStatus() should
return a different value and print "Change" to the PC, but this does not happen and the message is only printed when I press the button connected to pin 1.5

Can anyone advise me what I am doing wrong?

DeqingSun, 2024-05-24:
This feature was not tested because I develop the code on a Mac, and Mac have separate LED Status for keyboard. 

You have done nothing wrong, I need to check how to fix it.

Archetix, 2024-05-24:
Thank you very much for your prompt reply.

Archetix, 2024-05-25:
Hello.

Maybe this example for Arduino might help you.
https://github.com/arduino/Arduino/files/4949367/USBHID_LedStatus.zip

Here is a sample from HID.cpp:

if (requestType == REQUEST_HOSTTODEVICE_CLASS_INTERFACE)
	{
		if (request == HID_SET_PROTOCOL) {
			// The USB Host tells us if we are in boot or report mode.
			// This only works with a real boot compatible device.
			protocol = setup.wValueL;
			return true;
		}
		if (request == HID_SET_IDLE) {
			idle = setup.wValueL;
			return true;
		}
		if (request == HID_SET_REPORT)
		{
			if (setup.wLength == 2)
			{
				uint8_t data[2];
				if (2 == USB_RecvControl(data, 2)) 
				{
					_keyboardLedsStatus = data[1];
					return true;
				}
			}
			//uint8_t reportID = setup.wValueL;
			//uint16_t length = setup.wLength;
			//uint8_t data[length];
			// Make sure to not read more data than USB_EP_SIZE.
			// You can read multiple times through a loop.
			// The first byte (may!) contain the reportID on a multreport.
			//USB_RecvControl(data, length);
		}
	}

	return false;
}

uint8_t HID_::getKeyboardLedsStatus(void)
{
	return _keyboardLedsStatus;
}

Archetix, 2024-05-25:
Hello.

I think I've figured it out, just need to tweak it a bit.
These are my changes for USBhandler.c :

void USB_EP0_OUT() {
  if ((SetupReq == HID_SET_REPORT)) {
    // does not care led status for now
    LedStatus8_t = Ep0Buffer[0]; //by Archetix
}
  UEP0_T_LEN = 0;
  UEP0_CTRL ^= bUEP_R_TOG;
}

#pragma save
#pragma nooverlay

// by Archetix
uint8_t LEDStatus(void) {
  return LedStatus8_t; // The only info we gets
}

void USBInterrupt(void) { // inline not really working in multiple files in SDCC
  if (UIF_TRANSFER) {
    // Dispatch to service functions
    __data uint8_t callIndex = USB_INT_ST & MASK_UIS_ENDP;
    switch (USB_INT_ST & MASK_UIS_TOKEN) {
    case UIS_TOKEN_OUT: { // SDCC will take IRAM if array of function pointer is  
                          // used.
 
   USB_EP0_OUT(); //by Archetix

Mr-ShiYi, 2024-07-16:
请问NUM LOCK, CAPS LOCK or SCROLL LOCK对应window的返回值是什么

DeqingSun, 2024-07-27:
fixed with
https://github.com/DeqingSun/ch55xduino/commit/9752656ae3ab8bc945ef54cf7464c5126e5ba419

closed 2024-07-27 by DeqingSun