DeqingSun/ch55xduino #168: Add a function to the keyboard example to write strings (Not an issue)

NoNamedCat, 2024-06-02:
I'm making a macro keypad using the CH55xDuino and a simple board that has a ch552 chip.

I don't know if this could help anyone but i think that this would make a grat addition to the example for using the library:

```c++

#ifndef USER_USB_RAM
#error "This example needs to be compiled with a USER USB setting"
#endif

#include "src/userUsbHidKeyboard/USBHIDKeyboard.h"

unsigned char buttonPins[12]= {16, 15, 14, 32, 30, 31, 10, 11, 34, 33, 35, 13}; 
bool buttonPressPrev[12]={0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0};

void enterBootloader(){
  USB_CTRL = 0;
  EA = 0;                     //Disabling all interrupts is required.  
  TMOD = 0;                                                            
  __asm__ ("lcall #0x3800");  //Jump to bootloader code
}
void Keyboard_writeWord(const char *word) {
    // Recorrer cada carácter de la palabra
    for (int i = 0; word[i] != '\0'; i++) {
        // Escribir cada carácter utilizando Keyboard_write
        Keyboard_write(word[i]);
    }
}

void keyFunction(int function, bool keyUpDown){
  switch ( function )
  {
    case 0:
        if (keyUpDown){
          Keyboard_writeWord("sXMCg[9a'jb1[ux51}PD.{xo._EPd=EnuW4duMHSWtsoE=tT9b");
        }
        else{
        }
        break;
    case 1:
        if (keyUpDown){
          Keyboard_press('1');
        }
        else{
          Keyboard_release('1');
        }
        break;
    case 2:
        if (keyUpDown){
          Keyboard_press('2');
        }
        else{
          Keyboard_release('2');
        }
        break;
    case 3:
        if (keyUpDown){
          Keyboard_press('3');
        }
        else{
          Keyboard_release('3');
        }
        break;
    case 4:
        if (keyUpDown){
          Keyboard_press('4');
        }
        else{
          Keyboard_release('4');
        }
        break;
    case 5:
        if (keyUpDown){
          Keyboard_press('5');
        }
        else{
          Keyboard_release('5');
        }
        break;
    case 6:
        if (keyUpDown){
          Keyboard_press('6');
        }
        else{
          Keyboard_release('6');
        }
        break;
    case 7:
        if (keyUpDown){
          Keyboard_press('7');
        }
        else{
          Keyboard_release('7');
        }
        break;
    case 8:
        if (keyUpDown){
          Keyboard_press('8');
        }
        else{
          Keyboard_release('8');
        }
        break;
    case 9:
        if (keyUpDown){
          Keyboard_press('9');
        }
        else{
          Keyboard_release('9');
        }
        break;
    case 10:
        if (keyUpDown){
          Keyboard_press('a');
        }
        else{
          Keyboard_release('a');
        }
        break;
    case 11:
        if (keyUpDown){
          Keyboard_press('b');
        }
        else{
          Keyboard_release('B');
        }
        break; 
    default:
        break;
  }
}
void setup() {
  for(int i = 0; i<12 ; i++){
    pinMode(buttonPins[i], INPUT_PULLUP);
  }
  if(!digitalRead(buttonPins[0])){
    enterBootloader();
  }
  USBInit();
}

void loop() {
  bool buttonPress[12]={0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0};
  for(int i = 0; i<12 ; i++){
    buttonPress[i] = !digitalRead(buttonPins[i]);
    if (buttonPressPrev[i] != buttonPress[i]) {
      buttonPressPrev[i] = buttonPress[i];
      if (buttonPress[i]) {
        keyFunction(i, 1);
        delay(2);
      } 
      else {
        keyFunction(i, 0);
        delay(2);
      }
    }    
  }
}
```

DeqingSun, 2024-06-03:
This is a good example, thanks. I do not have have an example to make a macro keyboard because the DataFlash is too small to make a reconfigurable macro keyboard. It was only possible to make a reconfigurable keyboard, and "qmkCompatibleKeyboard" does that.

It would also be better if you can change the comment from Spanish to English.

NoNamedCat, 2024-06-03:
Do you think the Keyboard_writeWord function should be inside USBHIDKeyboard, or is it preferable to keep it separate in a different example?

DeqingSun, 2024-06-03:
I'll create a Keyboard_writeString next time I maintain the repo. I need to add it to multiple examples.

DeqingSun, 2024-07-27:
added in https://github.com/DeqingSun/ch55xduino/commit/5c67a9b4150ba4d17dfd99c5b2fc1f7bdf4f8a1a

closed 2024-07-27 by DeqingSun