DeqingSun/ch55xduino #171: Are the links outdated?

JmZeroQAQ, 2024-07-26:
I used two different links to install the support package, but both failed.

https://raw.githubusercontent.com/DeqingSun/ch55xduino/ch55xduino/package_ch55xduino_mcs51_index.json
**output**
```
Some indexes could not be updated. Get "https://raw.githubusercontent.com/DeqingSun/ch55xduino/ch55xduino/package_ch55xduino_mcs51_index.json": dial tcp: lookup raw.githubusercontent.com: getaddrinfow: The requested name is valid, but no data of the requested type was found.: https://raw.githubusercontent.com/DeqingSun/ch55xduino/ch55xduino/package_ch55xduino_mcs51_index.json
```

https://ch55xduino.pek3b.qingstor.com/package_ch55xduino_mcs51_newest_qingcloud_index.json

**output**
```
Some indexes could not be updated. Server responded with: 403 Forbidden: https://ch55xduino.pek3b.qingstor.com/package_ch55xduino_mcs51_newest_qingcloud_index.json
```

DeqingSun, 2024-07-26:
The first link worked for me. The second link does work because QingStor ended service to individual users. 

I'll move the packages to cloudflare COS this weekend.

JmZeroQAQ, 2024-07-26:
Thank you for your reply. I have resolved the issue. By setting up the proxy in the Arduino, I sucessfully install the package now.

closed 2024-07-26 by JmZeroQAQ