DeqingSun/ch55xduino #172: CH552: Unknown USB after downloading firmware

Bruur-vr, 2024-11-25:
Hello,

Before downloading this firmware the CH552 showed up as 4 different HID devices, I did not like the firmware it came with (I had no way of changing buttons) so I tried overwriting it through ArduinoIDE. I had no problem uploading when CH552 was in bootloader mode.

I adjusted the config.h since the premade board had the switches and encoder wired differently. Also it only has 6NeoPixels.
When first connecting (with old firmware) usbview said that max load was 100mA so I changed that too.

I installed everything in ArduinoIDE, selected board and adjusted parameters under "tools". Then hit upload while it was in bootloader mode. There was something about running "make flash" but that did not work for me: I also thought you either do it through ArduinoIDE or through makefile, right?

Since the upload it shows as Unknown USB device.
[image]

"Windows has stopped this device because it has reported problems. (Code 43)"

Problem code is 0000002B

Events says this:
```
Device USB\VID_0000&PID_0005\5&38e97a59&0&7 was not migrated due to partial or ambiguous match.

Last Device Instance Id: USB\VID_0000&PID_0003\6&31b1809d&0&1
Class Guid: {36fc9e60-c465-11cf-8056-444553540000}
Location Path: 
Migration Rank: 0xF000FFFFFFFFF100
Present: false
Status: 0xC0000719
```

in usbview it shows up like this:
[image]

Zadig shows up as it having no driver when out of bootloader mode.
[image]

When in bootloader it shows this, after I tried reïnstalling drivers (WINusb & libusb) but it makes no difference.
[image]

I am not well versed in these kind of topics so forgive me for not understanding in the first try.

Bruur-vr, 2024-11-25:
I am sorry, I posted this to the wrong github page.

DeqingSun, 2024-11-27:
It seems:
You can enter bootloader mode, and flash a firmware.
You can flash a stock firmware from the vendor, it works.
So there seems to be no hardware problem. The most possible reason is you compiled your firmware to run at 24MHz, while your voltage for CH552 is only 3.3V. 
The CH552-MacroPad-plus compiled the firmware for 16MHz. 
Confirm the voltage of VCC on CH552, then use drop down menu to choose 16M speed if the voltage is 3.3V.

DeqingSun, 2024-11-27:
The example does not work on my setup either, I got the same error. 
Let me check it.

DeqingSun, 2024-11-27:
Downgrade your Arduino to 0.0.16, which is the version that [wagiminator](https://github.com/wagiminator) prepared the code.

Bruur-vr, 2024-11-27:
> Downgrade your Arduino to 0.0.16, which is the version that [wagiminator](https://github.com/wagiminator) prepared the code. It worked for me.

Thanks for the advice, I downloaded the 0016 IDE, this IDE has no board manager where do I add the CH55x board?

DeqingSun, 2024-11-28:
Sorry I mean downgrade CH55xduino to 0.0.16

closed 2024-11-28 by Bruur-vr