DeqingSun/ch55xduino #177: Problems with PowerDelivery example

fitter22, 2025-01-18:
Hi, I decided to build the `PowerDelivery` example circuit and test it. It didnt work, there was a bit of interesting journey when I was learning the CH552 while trying to make it work, but lets cut to the chase here. Please correct me if I am wrong, because I am still quite new on these chips.

Seems like the updated bootloader (all ICs I have come with v2.5) came with few changes that made this example not work out of the box and can be a bit troublesome to understand why. My build worked with 5V USB, but after connecting it to USB PD, the charger either didnt start with it, or it was going in ON/OFF cycle every second or so. After ruling out problems with my HW, I got my attention into fact that MCU was going into download mode every time I connected it to my computer (I didnt have to press the boot button). I made a theory that, after connecting to USB PD charger, MCU was also booting to download mode and waiting, so charger was cutting it off from power.

When I was plying with WCH tool `WCHISPTOOL` on windows, I saw an option called `Enable NoKey serial port download` enabled by default. I disabled it, loaded hex made by Arduino IDE and... it worked! I have fully functioning USB PD sink, based on CH552.

Do you know if the same thing can be achieved just by using Arduino IDE? For now I think that PowerDelivery code will work only if MCU is flashed by that windows tool.

PS. I am using the P3.3 LED as "Power OK" indicator, telling that requested voltage was obtained successfully. It might be useful enhancement

DeqingSun, 2025-01-19:
Try this. You can set bootloader trigger pin in Arduino IDE.

[image]

fitter22, 2025-01-20:
Thanks.
I think it will also solve the problem, but I will not test it now - I did not expose the P1.5 to trigger the bootloader...

Although my guess is that not the charger, but new bootloader is causing the issues. My setup works with USB PD charger and bootloader set to P3.6. I just had to flash the FW using WCH tool and disable the `Enable NoKey serial port download`.
When I flash the same thing from Arduino IDE, it does not work

fitter22, 2025-01-27:
I will close it hence there is nothing that really should be changed.

closed 2025-01-27 by fitter22