DeqingSun/ch55xduino #178: vnproch55x does not work with default CH375 driver

Rybec, 2025-01-20:
As said in the title, vnproch55x does not seem to work with the default CH375 driver used by Windows 10 and Windows 11 (I've tested both).  I tested this from Arduino on Windows 11, and I tested it from the command line on Windows 10.  It doesn't matter if the device is put into bootloader mode or not, vnproch55x cannot find the device.

After using Zadig to replace the CH375 driver with libusb-win32, uploading from the command line on my Windows 10 machine with vnproch55x works as expected.

I'm using a Python script I wrote to enter bootloader mode by opening a serial connection at 1,200 baud, and I just tested it with a .hex file generated by ch55xduino in Arduino on another machine.  (I still haven't figured out the space in file paths issue, so I can't compile on my Windows 10 machine.)

DeqingSun, 2025-01-21:
When you can get WCHISPStudio official software working (CH375 driver).
Make the chip enter the bootloader and run the vnproch55x command. What is the console output? Also what is your device manager device looks like? You have to do it quickly as you only have 5 second before the bootloader timeout.

I want to check the output of these code.
https://github.com/DeqingSun/vnproch551/blob/c07331616c9802a1ca1510dde47b5e09261b8660/main.cpp#L335-L359

Rybec, 2025-01-21:
I'll try that tomorrow.  I installed libusb-win32 yesterday, and it works, but I can uninstall it and revert back to test this.  I did have WCHISPStudio working prior to installing libusb-win32.

The things I remember:
- vnproch55x times out, even if I set the timeout to 16+ seconds.
- The device showed up in Device Manager, mapped to COM6, when not in bootloader mode.
- When I put it into bootloader mode, it didn't show up at all, _but_ Device Manager can take well over 5 seconds to update when a new device is added, so I suspected that it wasn't showing up because it went out of bootloader mode too fast.  Maybe I was wrong and the CH375 driver wasn't installed until I installed WCHISPStudio?

Anyhow, it's way too late for me to do right now, but tomorrow I'll uninstall libusb-win32 and get you that output.

If it just turns out there was no bootloader driver installed, that's trivial to solve, and I can just let the people at Adafruit know they need to add driver installation to their setup tutorial for the board I'm using.

Rybec, 2025-01-22:
To uninstall libusb-win32, you really have to be fast.  If don't check the box to delete the driver and press the button before it leaves bootloader mode, it will silently fail to remove the driver.  It took me four tries!

Ok, here's what I've got:
1. I removed libusb-win32, then I reinstalled WCHISPStudio.
2. I attempted to upload with vnproch55x, and it was successful.
3. I removed the driver installed by WCHISPStudio, upload was unsuccessful.  (I did this mainly as a sanity check.)
4. I reinstalled WCHISPStudio again, which reinstalled the default driver, and upload is once again successful.

So, as far as I can tell, the problem is not vnproch55x but just lack of information.  My windows did _not_ automatically install the bootloader driver (and neither did my son's school Windows laptop that I tried it on), and I never came across any information suggesting that installing WCHISPStudio would install a working driver.  I only installed it as a last ditch effort, in hopes that maybe _something_ would work, and I was not aware that this installed the bootloader driver.

So the best fix for this might just be to update the readme for this project to mention that if Windows doesn't automatically install the driver, you can do it manually by downloading and installing WCHISPStudio.  (Alternatively, and I just tested this, you can also install the libusb-win32 driver _without_ having the CH375 driver already installed, with Zadig by plugging the device in in bootloader mode and looking for "Unknown Device #1" or something similar, but doing it this way requires changing the righthand driver field to libusb-win32, which makes it even harder to get within the bootloader timeout.  (It only took me two tries, but I already knew what to expect.)

Anyhow, I'm going to document all of this and send to Adafruit, in hopes that they will include this information on the Learn page for the QT Py CH552, but it might also be a good idea to mention this in the documentation for this project as well!

Thanks for helping me to get to the right place in figuring this out.  This has been around two weeks of frustration, and figuring out exactly why it wasn't working is really nice, because now I don't have to rely on fixes where I don't know how they are fixing the problem.

I'm going to close this, since the actual problem wasn't what I put in the title, _and_ the problem is solved.

closed 2025-01-22 by Rybec

Rybec, 2025-01-23:
Here's a standalone driver installer from WCH that includes the CH375 driver.  This might be a better option for those who don't want WCHISPStudio.

https://www.wch-ic.com/downloads/CH372DRV_EXE.html

DeqingSun, 2025-01-23:
That is different from my memory. However there is no way for me to test it.

I remember on Win10 the windows automatically search for the driver from Microsoft, and installed it. Although I can not find the post on wch support bbs, I remember WCH registered their drivers on Microsoft. 

One big difference I can imagine, is that when I test it several years ago, the bootloader does not have time out and windows have a long time to install the driver. But now it is possible that windows can not finish in 5 seconds and just give up every time. Maybe a erased chip can stay in bootloader long enough for windows to install driver.

Rybec, 2025-01-23:
This is very possible.  Windows can be quite finicky about timing on stuff like this.  Even when trying to uninstall the driver, if the bootloader times out before it is done uninstalling, it will silently fail and the driver will still be there.  And Zadig had a similar issue, though about 50% of the time, it would tell me it had failed.

Maybe just say that if Windows fails to install the driver automatically, you might have to install it manually using that vendor link.  That way if people have an older chip that doesn't time out super fast, they don't think they have to install the driver manually anyway, but if they have a newer chip like the one I have, they aren't left spending almost two weeks trying to figure out why it isn't working.

DeqingSun, 2025-01-23:
Make sense, Readme updated.