DeqingSun/ch55xduino #179: Cycles Listed in  WS2812 Assembly Don't Match Datasheet

Rybec, 2025-02-12:
This is more of a question than a real issue.  I can't find any CH552 documentation from WCH on cycles per instruction.  (Or really any detailed instruction set information.)  In the WS2812 code generated from the Python script, the inline assembly has comments after lines where timing is important noting the length in bytes of each instruction as well as the number of cycles it takes.  Except, those numbers appear to be wrong.  The number of bytes appears to be correct but in some cases the number of cycles don't match the numbers found in 8051 datasheets.

Following are my sources:

https://ww1.microchip.com/downloads/en/DeviceDoc/doc0509.pdf
https://developer.arm.com/documentation/101655/0961/8051-Instruction-Set-Manual/Instructions

These appear to be in agreement.  Two examples, found in optionalLink_WS2812_P1_0.c, on lines 69 and 71.  69 is a SETB instruction with a  bit address.  According to both of those documents this instruction is 2 bytes but only takes 1 cycle.  Similarly, 71 is a CLR instruction with a bit address, which again is 2 bytes but only takes 1 cycle.  In both of those cases, they are listed as 2 bytes and 2 cycles.

Now, even if the numbers in the code comments are wrong, this is clearly not a huge timing issue, given that this code works perfectly fine on both of my QT Py CH552s.  But, my question is: Do you have access to official CH552 documentation that indicates these instructions take 2 cycles?  Both of my sources are generic 8051 data sheets/information, that conceivably could be different from the CH552 implementation of the 8051 hardware.  If you have documentation listing the number of cycles for each instruction for the CH552 that are different from the generic data sheet and instruction listing above, I would really like to know where I can get a copy.  In the near future I plan to start writing some drivers for the CH552, some of which may require precision timing, and if the generic information is wrong, it would be really handy to have the correct information.

There are also some branch instructions with confusing cycle number listings.  I understand the DJNZ instruction on line 76 in the same file, where the cycle length is listed as `2/4|5|6`, given that the instruction can jump to itself multiple times before moving on (though, the datasheet says it takes 2 cycles, so the 5 option shouldn't be impossible, as it should always be a multiple of 2).  A few lines before though, on line 70, there's a JC instruction with cycles listed as `2/4|5`, which jumps forward (skipping a CLR instruction) when the condition is met.

Anyhow, do you know what's going on here?  Or do I need to talk to Stephen Erisman, who according to the comments at the beginning is responsible for a major rewrite of this module?

(On a side note, are you alright with me posting this kind of thing as an issue here, or should I contact you via the email address listed in the source code files instead?)

DeqingSun, 2025-02-12:
Please do not use the word that scary in the title. You are not comparing the CH552 code with CH552 document. You are comparing with some other documents.

Any way the document is on, and the title is "CH55X 汇编指令说明" in case the link fails in the future.
https://www.wch.cn/uploads/file/20210617/1623896732811060.pdf

Basically the cycles are the same to the bytes for most instructions. And successful branching takes generally 2 more cycles. Due to some pipeline alignment mechanism that WCH does not want to explain, any flash access with odd address get extra penalty, both for command on odd address, and jump to an odd address. 

The odd/even address is important in some cases such as delayMicroseconds, power-delivery decoding, etc. There is ```.even``` keyword and [CSEG are converted to even number](https://github.com/DeqingSun/ch55xduino/commit/255de23448bff76a9a460a12d0f893bd069a4861) to ensure the functions are linked at even address.

Here is some translation, I did proof-reading and the translation is accurate:

```
CH55X Assembly Instructions
Applicable models include: CH551, CH552, CH554, CH558, CH559
CH55X assembly instruction cycle overview:
a.The number of instruction cycles for non-skip instructions is the same as the number of instruction bytes (except for DIV instructions in CH551/2/3/4);
b.Most instructions are single-byte, 1T single-cycle instructions;
c.Jump instructions containing MOVC/RET/CALL usually have a number of cycles more than the number of instruction bytes;
d.RET/RETI instructions start at 3 more cycles, plus 1 cycle if the return address is odd;
e.MOVC instruction is 4 cycles more, such as the address of this instruction is odd plus 1 cycle, such as the address of the next instruction is odd plus 1 cycle;
f.JMP@A instruction 2 more cycles from, such as this instruction address is odd plus 1 cycle, such as the target address is odd plus 1 cycle;
g.JB/JNB/JBC/CJNE_A,dir/DJNZ instruction is 2 cycles more, if the address of this instruction is odd then add 1 cycle, if the target address is odd then add 1 cycle.
If the address of this instruction is odd, then add 1 cycle, if the target address is odd, then add 1 cycle;
h. The rest of the jump instructions are 2 cycles more, such as the target address is odd plus 1 cycle;
i. If no jump occurs in the conditional jump instruction, the number of cycles is the same as the number of bytes in the instruction.
Note: The above cycles refer to the inverse of the current CH55X system main frequency.

Translated with DeepL.com (free version)
```

serisman, 2025-02-12:
Yes, as DeqingSun mentioned, the CH55x MCUs DO use the 8051 instruction set, but are HIGHLY optimized to use significantly fewer clock cycles per instruction, and to run at much higher clock frequencies.  Most instructions are 1T except for those involving IO/Mem access or branching.  As DeqingSun mentioned, most instructions take the same number of cycles as they are bytes long.  Looks like DeqingSun already pointed you at the datasheet that shows cycles per instruction.

As for why the WS2812 code doesn't necessarily match the timing found in its datasheet(s), that is because of research done by 'Tim':

* https://cpldcpu.wordpress.com/2014/01/14/light_ws2812-library-v2-0-part-i-understanding-the-ws2812/
* https://wp.josh.com/2014/05/13/ws2812-neopixels-are-not-so-finicky-once-you-get-to-know-them/

Through some impressive investigation he found out what they are actually doing internally and came up with a more reliable and faster way to handle controlling them.  All I did was bring his concepts over to this project in order to be able to support more clock frequencies as well as fix some timing bugs I was seeing.

Rybec, 2025-02-12:
Ok, so first, I apologize for the title.  I didn't intend for it to be scary.  If I'm just asking a question in the future, I'll note it in the title.  (I also realize now the ambiguity in the title, as it could also have referred to the NeoPixel specifications, which is not what I meant.)

Second, thank you so much for the information!  I spent several hours last week trying to find a more complete datasheet than the one on the English version of the WHC website with no luck.  The English one does not contain any information about the instructions at all.  I also appreciate the translation, as I don't read Chinese.  I also see that the datasheet you provided includes the new instruction mentioned in the promotional material, which I also spent a while last week trying to find information on, again with no luck.

Anyhow, I very much appreciate the responses from both of you.  I'm trying to put together some learning resources for this chip, including assembly and C tutorials, and I've ran into quite a few roadblocks.  This solves what I hope are the last two issues.  Given the complexity of branch instruction timing, maybe the NeoPixel is a poor choice for the first driver writing tutorial...

Thanks again for the help!  (And thank you both for the work you done supporting this chip.)

closed 2025-02-12 by Rybec