DeqingSun/ch55xduino #180: How to solve This Error: Cannot perform port reset: 1200-bps touch: opening port at 1200bps: Invalid serial port

BB-KING777, 2025-02-26:
I am currently building a microcontroller board using CH552G. after updating to libusb-win32, I wrote the USB CDC firmware that exists in the Sample code in the Arduino IDE and changed the driver to USB CDC in Zading. Then I was able to confirm that it was recognized as shown in image 1.

[image]

Next, I tried to write the Blink code in the Arduino IDE,

```
Performing 1200-bps touch reset on serial port COM8
Cannot perform port reset: 1200-bps touch: opening port at 1200bps: Invalid serial port
"C:\Users\syuta\AppData\Local\Arduino15\packages\CH55xDuino\tools\MCS51Tools\2023.10.10/win/vnproch55x" "C:\Users\syuta\AppData\Local\arduino\sketches\4C3BDA776A3DD2D9DE253379A517E1E5/Blink.ino.hex" -r2
------------------------------------------------------------------
CH55x Programmer by Deqing
Updated on: 2023/10/10
------------------------------------------------------------------
Load file as hex
   Loaded 4524 bytes between: 0000 to 11F9
ch375Version 35
CH375GetUsbID 0
Found no CH55x USB
Failed uploading: uploading error: exit status 1
```

I get the following error message

I can't figure it out no matter how many times I try. Also, USB is not recognized unless RST is set to GND. This is probably normal, right?

Please let me know if I am doing something wrong or failing.

Maybe I am just new to information technology and don't know anything about it.

BB-KING777, 2025-02-26:
SORRY, I close this issue myself.

closed 2025-02-26 by BB-KING777

DeqingSun, 2025-02-26:
I'm not sure what you solved. But here is some suggestions based on the information in your post

Use 10K-20K resistor to pull up D+. 1K is a too small resistor and it will corrupt USB connection.

You do not need to touch or connect RST in most cases.

I'm not sure why you get the small/large model link issue, please update the repo to a new version as the model was switched to large a long time ago.