DeqingSun/ch55xduino #181: is USBCDC blocking or not?

rew1ndustry, 2025-04-24:
after completing programming, i disconnected the USB lines, because they are not required in my project, which runs on 3v3.

suddenly nothing is working and, since i left the USBSerial_print() calls in my code, i wonder if these calls are blocking, when there is no USB connected?

DeqingSun, 2025-04-24:
It is more like you use the 24M clock for 3.3V system. And you have to use 16M or lower. You can find settings in menu.

rew1ndustry, 2025-04-24:
thank you very much, i doubt i would have ever noticed the clock problem, if you had not pointed this out.

the chip is now working, and i can confirm that USBSerial does not block when there is no USB connection.