DeqingSun/ch55xduino #182: Request for Code Example: CH552 USB Keyboard Host with ASCII Output and Key Press INT

Mostez, 2025-04-26:
Hello, and thank you for the fantastic CH55xduino project!

I’m currently working on a project where I aim to use a CH552 microcontroller as a USB keyboard host. The goal is to output the pressed key as ASCII characters on an IO pin at 9600 baud. Additionally, I’d like to implement an interrupt (INT) where a designated output pin goes high (or low) whenever a key press is detected.

If possible, could you provide guidance or example code demonstrating this functionality? Specifically:
How to configure the CH552 as a USB host to handle keyboard input.
Sending the pressed key’s ASCII value as UART output at 9600 baud.
Setting up an INT pin to signal key press events.

I believe this example would be very helpful for others in the community working on similar projects. 

Thank you for your time and support!

DeqingSun, 2025-04-26:
You should read the datasheet before you ask a question like this.

closed 2025-04-26 by DeqingSun

Mostez, 2025-04-27:
Thank you for your reply and advice. I understand the importance of reviewing datasheets thoroughly, and I appreciate your suggestion. However, I’m tied to a tight deadline and currently lack the time to delve deeply into the datasheet. Additionally, I’m quite new to working with CH55x microcontrollers and still navigating the learning curve. Unfortunately, local market options are limited, which makes this chip one of the few viable choices for my project.

I have access to CH554, CH559, and CH549 microcontrollers. Considering the constraints of size and price, CH554 and CH549 are the best choices for my project, making CH554 the most practical option given its USB host capabilities.

Thank you